underground-software/RPi.GPIO2 #3: Pulse with Modulation (PwM) does not work

theyoyojo, 2019-11-22:
This will probably require modifications to the kernel

lurch, 2020-03-02:
The 'W' in PWM stands for "Width" ;-)

The Raspberry Pi supports hardware PWM on a limited number of pins (see e.g. https://pinout.xyz/ ), but RPi.GPIO doesn't support the Pi's PWM hardware. Instead RPi.GPIO allows you to use software PWM on any pin - the downside being that software PWM can be quite jittery.

theyoyojo, 2020-03-03:
Thank you for your correction of my typo.

I have discussed this issue with @Codom and @Maickii and we have decided to extend this library to support hardware PwM via the sysfs interface exposed by the kernel and to experimentally implement software PwM via libgpiod for comparison with the performance of RPi.GPIO's direct software PwM.

I suspect that the performance will be worse but we will see.

lurch, 2020-03-04:
> I suspect that the performance will be worse but we will see.

Yeah, the software PWM in RPi.GPIO is in C, whereas I guess the software PWM you're adding here will be in Python?

theyoyojo, 2020-03-04:
Our library is indeed entirely written in python, but I don't think that will be the main reason for the relative degradation in performance of software PwM. Since we rely on libgpiod for the underlying interaction with the kernel, and since libgpiod makes `ioctl(2)` syscalls every time it updates the value of a pin, I suspect the performance difference will be due to the overhead difference between that method and RPi.GPIO's writes to the memory-mapped GPIO registers. However, we also plan to transparently support hardware PwM much as possible.

Of course, we still need to actually test this.

@lurch what do you think if this theory?

lurch, 2020-03-04:
:man_shrugging: I dunno.

> Of course, we still need to actually test this.

Yeah, that's probably the way to go.

closed 2020-06-19 by theyoyojo