underground-software/RPi.GPIO2 #10: Support for use as a drop-in replacement for RPi.GPIO for gpiozero

theyoyojo, 2020-05-08:
In response to: https://github.com/gpiozero/gpiozero/issues/840

bennuttall, 2020-05-14:
In particular, to be a drop-in replacement, there would need to be a "pin factory" implementation, which normalises the functionality of a pin library into a standard set of methods. The RPi.GPIO one is here: https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/rpigpio.py

More info: https://gpiozero.readthedocs.io/en/latest/api_pins.html

theyoyojo, 2020-05-15:
A test-driven way we could approach this feature would be to integrate a version of that file into our testing system as we move towards API-equivalence with RPi.GPIO. @waltdisgrace what do you think?

theyoyojo, 2020-06-12:
I think that we are getting close to full compatibility with #23 and #25

lurch, 2020-06-12:
Nice, I'll try to have a look at it this weekend (if I remember!) :+1:

theyoyojo, 2020-06-19:
Oops, looks like I forgot another constant:

>>> import gpiozero.pins.rpigpio as rpigpio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/rpi/python3-libgpiod-rpi/lib/python3.7/site-packages/gpiozero/pins/rpigpio.py", line 91, in <module>
    class RPiGPIOPin(LocalPiPin):
  File "/home/rpi/python3-libgpiod-rpi/lib/python3.7/site-packages/gpiozero/pins/rpigpio.py", line 104, in RPiGPIOPin
    'serial':  GPIO.SERIAL,
AttributeError: module 'RPi.GPIO' has no attribute 'SERIAL'

theyoyojo, 2020-06-19:
Alright it imports correctly now, but gpiozero now throws the following when I try to evaluate PiFactory.pi_info()
```
>>> import gpiozero.pins.rpigpio as rpigpio
>>> fac = rpigpio.LocalPiFactory()
>>> fac.pi_info
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/rpi/.local/lib/python3.7/site-packages/gpiozero/pins/__init__.py", line 188, in <lambda>
    lambda self: self._get_pi_info(),
  File "/home/rpi/.local/lib/python3.7/site-packages/gpiozero/pins/pi.py", line 120, in _get_pi_info
    self._info = pi_info(self._get_revision())
  File "/home/rpi/.local/lib/python3.7/site-packages/gpiozero/pins/local.py", line 109, in _get_revision
    raise PinUnknownPi('unable to locate Pi revision in /proc/device-tree or /proc/cpuinfo')
gpiozero.exc.PinUnknownPi: unable to locate Pi revision in /proc/device-tree or /proc/cpuinfo
```

bennuttall, 2020-06-19:
Are you able to get the revision code another way? We ought to patch that. @lurch any ideas for doing that outside of Raspbian?

theyoyojo, 2020-06-19:
mistakenly closed instead of #37

theyoyojo, 2020-06-19:
About to look into revision code issue. Just released alpha2 to pypi so people can `pip install` a version that is importable by gpiozero

theyoyojo, 2020-06-19:
On one of the gpiozero issues: https://github.com/joan2937/pigpio/issues/340,
@guymcswain links to the following hack: https://www.raspberrypi.org/forums/viewtopic.php?p=1428987#p1428987

theyoyojo, 2020-06-19:
One way to do it is to have a table mapping values of `cat /proc/device-tree/model` to revision strings

theyoyojo, 2020-06-19:
Using this patch to gpiozero, I was able to get gpiozerocli.pinout.main() to work with RPi.GPIO2 0.3.0a2

PR #38 (2020-06-19): Expose mistakenly unexposed RPi.GPIO.SERIAL constant
This enables exception-free transparent execution of `import gpiozero.pins.rpigpio as rpigpio`

fixes #10