underground-software/RPi.GPIO2 #49: The initial arguments defined in setup function does not take effect

vava24680, 2020-08-11:
Hi, I had the problem that the `initial` argument defined in `setup` function does not take effect. Below is what I did.
```
from RPi import GPIO

GPIO.setmode(GPIO.BCM)
GPIO.setup(22, GPIO.OUT, initial=GPIO.LOW) # It does not set the initial value to low.
```
I ran this code under the newest version of this package and the platform is Raspberry Pi 4B+.

No matter I passed low or high, it always set the same initial value. Can somebody help me or point out what's going on?

Thank you

theyoyojo, 2020-08-12:
Hi,

Thanks for raising this issue, I'll look into it and see what I can find.

vava24680, 2020-08-14:
I think I may know what's the problem.
The value of `GPIO.LOW` is 2, so when it will be treated as the active high...

theyoyojo, 2020-08-14:
I was able to reproduce this issue. Definitely worth fixing.

I extended your code slightly to create this reproducer:

```python
#!/bin/python3

from RPi import GPIO
from RPi import GPIO_DEVEL

pin=22

GPIO.setmode(GPIO.BCM)
GPIO.setup(pin, GPIO.OUT, initial=GPIO.LOW) # It does not set the initial value to low.

input()

GPIO_DEVEL.Reset()

GPIO.setmode(GPIO.BCM)
GPIO.setup(pin, GPIO.OUT, initial=GPIO.HIGH) # Same behavior as initial=GPIO.LOW

input()
```

Would you like to submit a fix yourself? If not, I will patch the code.

vava24680, 2020-08-14:
I think this is a bug from the python binding of the libgpiod. This [link](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/bindings/python/gpiodmodule.c?h=v1.5.x#n75) is how the `GPIO.LOW` is defined in the binding. As far as I know how the enum works in the C, the value of `gpiod_ACTIVE_LOW` will be 2.

I think I can submit a fix for this strange behavior by hardcoding the value of `GPIO.LOW`  as 0 in the `core.py`. This is a temporary  solution, but I think it's ok for the current using. Anyway, I will try to understand whether my guess is correct and see if I can submit a patch to the libgpiod.

theyoyojo, 2020-10-07:
@vava24680 hey, are you still working on this?

vava24680, 2020-10-08:
Sorry for the delay, I think I may patch this bug on this weekend.